martin-ger/MQTT_VPN #2: doubt: accessing ports on the nodes

beckmx, 2019-08-06:
I was reading your readme file and was wondering if after getting an IP I can access whatever port on whatever protocol? for instance if I want to get the 80 port to see a page in the module and then switch to another port to see stream of images, can this be done?

martin-ger, 2019-08-07:
Yes, you can. You can access any port on the participating nodes. Without further software (routing), you cannot accesss nodes outside the VPN.

beckmx, 2019-08-07:
From what repository is this dependency?
`nacl/crypto_hash.h`

martin-ger, 2019-08-07:
Package libnacl-dev

beckmx, 2019-08-07:
wow, I just tested, it works like a charm, if it is not much to ask can you elaborate how it works behind the scenes in the ESP?

martin-ger, 2019-08-28:
I would guess, it is an ESP issue. Does this happen under high load? "tcp_receive: valid queue length" failed" sounds like a kind of overflow...

pastry777, 2022-01-04:
> вау, я только что протестировал, это работает как шарм, если нечего спрашивать, можете ли вы подробнее рассказать, как это работает за кулисами в ESP?

how did you manage to get this running on Linux? I installed libnacl-dev, but mqtt_vpn.c does not want to compile, the error persists:
/ usr / bin / ld: /tmp/ccs2jNoK.o: in the "msgarrvd" function:
mqtt_vpn.c :(. text + 0x8dc): undefined reference to "crypto_secretbox_xsalsa20poly1305_ref_open"
/ usr / bin / ld: /tmp/ccs2jNoK.o: in the "main" function:
mqtt_vpn.c :(. text + 0xfa0): undefined reference to "crypto_hash_sha512_ref"
/ usr / bin / ld: mqtt_vpn.c :(. text + 0x1874): undefined reference to "crypto_secretbox_xsalsa20poly1305_ref"
collect2: error: ld returned 1 exit status

I'm already tired of looking for a solution, was it really difficult for the author to write a couple of lines of what and how to install, many, including myself, do not understand how to do this = (

> wow, I just tested, it works like a charm, if it is not much to ask can you elaborate how it works behind the scenes in the ESP?

how did you manage to get this running on Linux? I installed libnacl-dev, but mqtt_vpn.c does not want to compile, the error persists:
/ usr / bin / ld: /tmp/ccs2jNoK.o: in the "msgarrvd" function:
mqtt_vpn.c :(. text + 0x8dc): undefined reference to "crypto_secretbox_xsalsa20poly1305_ref_open"
/ usr / bin / ld: /tmp/ccs2jNoK.o: in the "main" function:
mqtt_vpn.c :(. text + 0xfa0): undefined reference to "crypto_hash_sha512_ref"
/ usr / bin / ld: mqtt_vpn.c :(. text + 0x1874): undefined reference to "crypto_secretbox_xsalsa20poly1305_ref"
collect2: error: ld returned 1 exit status

I'm already tired of looking for a solution, was it really difficult for the author to write a couple of lines of what and how to install, many, including myself, do not understand how to do this = (

pastry777, 2022-01-04:
I managed to compile with the command:
gcc -o mqtt_vpn /mqtt_vpn.c -l paho-mqtt3c -l nacl