martin-ger/MQTT_VPN #13: how to run it on Linux?

pastry777, 2022-01-04:
You write "Requires the Paho MQTT C client library (https://www.eclipse.org/paho/files/mqttdoc/MQTTClient/html/index.html) and libnacl for encryption operations." Was it difficult to describe the installation process? I've been struggling for a whole day now trying to run your code on Ubuntu...

i am trying to execute:
gcc -o mqtt_vpn /mqtt_vpn.c -l paho-mqtt3c
but keep getting the error:
/ usr / bin / ld: /tmp/ccjrc57w.o: in the "msgarrvd" function:
mqtt_vpn.c :(. text + 0x8dc): undefined reference to "crypto_secretbox_xsalsa20poly1305_ref_open"
/ usr / bin / ld: /tmp/ccjrc57w.o: in the "main" function:
mqtt_vpn.c :(. text + 0xfa0): undefined reference to "crypto_hash_sha512_ref"
/ usr / bin / ld: mqtt_vpn.c :(. text + 0x1874): undefined reference to "crypto_secretbox_xsalsa20poly1305_ref"
collect2: error: ld returned 1 exit status

pastry777, 2022-01-04:
is it possible to write this code in python? it would be much easier with him ...

pastry777, 2022-01-04:
I managed to compile with the command:
gcc -o mqtt_vpn /mqtt_vpn.c -l paho-mqtt3c -l nacl

closed 2022-01-04 by pastry777

bwjohns4, 2022-11-18:
Here is a .sh script I made to fully build the Linux version. Before running it just change the extension from .txt to .sh and do:
`sudo chmod +x ./mqttVPNdependencyInstaller.sh`
then it should go through, download the git repos, build the libraries, install the libs, and run the make file.
[mqttVPNdependencyInstaller.txt](https://github.com/martin-ger/MQTT_VPN/files/10042015/mqttVPNdependencyInstaller.txt)