martin-ger/MQTT_VPN #19: Suggestion for having many clients(devices)

garudaonekh, 2023-01-15:
Hi,
I need suggestion for having multiple devices. Currently, I tested with two devices(10.0.1.2 and 10.1.3). The demon listen for data from response data from 10.0.1.1. What if I have 100 devices or 200 devices?

Please give me suggestion for this as I am building Esp32 sprinkler system depending on this library.

Pablo2048, 2023-01-15:
The client subscribe to only two topics IMHO - his own address and the broadcast one (https://github.com/martin-ger/MQTT_VPN/blob/6c2e21761bafff81b4302e523887d71a41bbfb51/mqtt_vpn_arduino/src/mqttif.cpp#L110 and the next one). You don't need any 'subtopic'...

garudaonekh, 2023-01-15:
> The client subscribe to only two topics IMHO - his own address and the broadcast one (
> 
> https://github.com/martin-ger/MQTT_VPN/blob/6c2e21761bafff81b4302e523887d71a41bbfb51/mqtt_vpn_arduino/src/mqttif.cpp#L110
> 
> and the next one). You don't need any 'subtopic'...

Hi, i am referring to the daemon.

Pablo2048, 2023-01-15:
The 'daemon' is just the same station as any ESP so it will subscribe to two topics: his own ip address and the broadcast one. If any station will communicate with the PC station it will publish to the PC station topic (his ip address) and awaits the reply (or replies) in its own topic (ip address).

garudaonekh, 2023-01-15:
> The 'daemon' is just the same station as any ESP so it will subscribe to two topics: his own ip address and the broadcast one. If any station will communicate with the PC station it will publish to the PC station topic (his ip address) and awaits the reply (or replies) in its own topic (ip address). See
> 
> https://github.com/martin-ger/MQTT_VPN/blob/6c2e21761bafff81b4302e523887d71a41bbfb51/mqtt_vpn.c#L336
> 
> and the next line...

I mean a single daemon with 100 ESP32 devices connected to it.

Pablo2048, 2023-01-15:
Then it's not about daemon, but used MQTT broker instead. Just to make it short: every MQTT VPN client (any ESP or PC) will made 2 subscriptions. Traffic is determined by clients and your application (how many times, how fast, how often they will communicate). The VPN does just simple thing - client 'packs' every ip datagram do data part of topic and the topic itself came from destination ip address - its just unicast (single) ip address, or broadcast topic address.

closed 2023-01-27 by garudaonekh